supermerill/SuperSlicer #4269: External lines protruding from others when using archene and external line width

GalvanicGlaze, 2024-05-22:
### What happened?

Weird external artifacts when using archene and suggested external line width on per model bases. Was fine in 2.5.59.3 but no good in 2.5.59.9/10

I'll work around it but though I'd would share just in case

[image]
[image]

### Project file & How to reproduce

[SB cover.zip](https://github.com/supermerill/SuperSlicer/files/15402090/SB.cover.zip)

### Version

2.5.59.10

### Operating system

Windows 10

### Printer model

Voron Trident

supermerill, 2024-05-22:
Seems like the overhang flow is just at the threshold. just increase it a little bit, or deactivate it.
Your project has fired many assert  checks, I use my entire day to fix most of the little issues it undercover (pesky unreliable arachne generator)

GalvanicGlaze, 2024-06-12:
> Seems like the overhang flow is just at the threshold. just increase it a little bit, or deactivate it. Your project has fired many assert checks, I use my entire day to fix most of the little issues it undercover (pesky unreliable arachne generator)

Hi, im not sure how this all works but though I would mention the problem is still currently in the latest RC and night_dev for me

supermerill, 2024-06-12:
increase the threshold for overhang flow.

GalvanicGlaze, 2024-06-12:
Yeah I know the work around, I thought it was something you were fixing so though it would be good to keep you updated. I changed mine to 100% permanently to work around it. Thanks

closed 2024-06-12 by supermerill