supermerill/SuperSlicer #4274: Seam Modifier positioning, XY replaced

CuredPrusa, 2024-05-24:
Version: 2.5.60.0
Build: SuperSlicer_2.5.60.0_2024-05-21
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19045), 64-bit edition

I have noticed replaced behaviour of coordinates for the Seam Sphere Position Modifier. I don't know if other modifiers have the same problem, I didn't check. 

Problem:
X coordinate moves north-south
Y coordinate moves east-west

Expected:
X coordinate moves east-west
Y coordinate moves north-south

[image]

supermerill, 2024-05-27:
Currently, the axis on the 3D view are that way, for everything.

Many printer have different real axis, so the 3D view as a 'classroom' axis set: if you look from the top, you have the x that goes from left to right, and the y from bottom to top.

CuredPrusa, 2024-05-27:
What do you mean for everything?
THe problem is:
if I move the object +10, it goes 10mm to the east.
But if I move the modifier +10, it goes 10mm to the north.
I think it should go to the east too.

CuredPrusa, 2024-05-27:
You see, the sphere is located here like X -10, Y -90.
But the instruction is X -90, Y -10. 

[image]

CuredPrusa, 2024-05-27:
And while we're at this screenshot, is there a way to remove xyz arrows from the plate. I should have my model in the middle of the plate, and in the middle of the model is a hole which I can not see because of an arrow. So I have to move model around to inspect it.

supermerill, 2024-05-27:
Note:
when you have multiple parts into one object.
 * each position & rotation of each part is from the "full object" position&rotation&scale
 * if you rotate your "full object" by 90° in the Z plane, then if you move a part by 10 x, you'll see it moving in the y direction because it's rotated by the object afterwards.
 
 The not nice stuff is that the rotation matrix is hidden after each rotation, maybe I should change this behavior.

CuredPrusa, 2024-05-27:
I think I understand what you're saying. Yes, the object was rotated. It would be nice to have the actual condition more transparent for better understanding. Or to simply make modifier follow the actual position of the object, regardless of the original.

supermerill, 2024-05-27:
> Or to simply make modifier follow the actual position of the object, regardless of the original.

I won't do it. If you rotate the object, you want all part to rotate together. If you want to rotate only a part, rotate only this part.

CuredPrusa, 2024-05-27:
ok. clear. thanks.