supermerill/SuperSlicer #4275: Estimate time of completion for sliced files

GalvanicGlaze, 2024-05-24:
Any chance we could get a line added under the total printing time for a sliced file to say what time it's expected to finish?

[image]

neophyl, 2024-05-24:
How ?  You could start the print of the gcode file at any point after its sliced.  The SLICER has no idea when that will be.  Could be  a minute, and hour, 3 weeks etc.  Just because its been sent to a printer that does not mean its going to start the print.

Finish time is a function of your printer firmware.  It knows when it started, it knows the total time to print and so if it has a real time clock then it can tell you when it should finish.

GalvanicGlaze, 2024-05-24:
> How ? You could start the print of the gcode file at any point after its sliced. The SLICER has no idea when that will be. Could be a minute, and hour, 3 weeks etc. Just because its been sent to a [...]

Yeah , I would hope common sense would prevail

I always slice my files and send them directly to the printer to print with the upload and print button in SS.
Everyday when I print and I want the print to end at a certain time so I can swap the plate out before bed or before I go to work I need to work out what time it would finish and then I add or remove parts to get the timing right. 

I want an option for it to take the current time and add the approximate print time and tell what time it would finish if I sent it now with a hot printer ready to go, same thing I have to work out myself currently.

krautech, 2024-05-25:
I agree, as simple of an addition this is, itd be great. I also upload + print directly to printer and having this would be great. I could easily see an ETA and know what time itd finish for the same reasons.

closed 2024-06-12 by supermerill