supermerill/SuperSlicer #4276: Setting: XY Compensation is reversed

madejackson, 2024-05-24:
**Bug description**
in 2.5.59.10, The Setting "XY Compensation" is reversed. See Printscreen for comparison with Version 2.5.59.6

**Expected behavior**
Extrusion / hole should be shrunk instead of expanded.

!!>> Project File <<!!
[D0_bearing_based_spirographic_WDT_tool_2024_14.zip](https://github.com/supermerill/SuperSlicer/files/15431013/D0_bearing_based_spirographic_WDT_tool_2024_14.zip)

**Screenshots**
Comparison between Versions:
[image]

**Desktop:**
 - OS: Windows
 - Version 2.5.59.10

CuredPrusa, 2024-05-24:
As SuperMeril explain to me once, "minus" means shaving off the perimeter material.  Inside or outside. 
And yes, this doesn't match the actual descriptive text (mouse hover).
And yes, some other versions reverse that for I don't know what reason. 

That's why this function is on my check list as one of important functions which have to be checked when testing a new version.
That's why I always save my projects with SS version number in project's name (file name). Until I find additional time and energy to test new SS version on old project, I use the version in which project was conceived.  It is tedious, but what can you do.  Guys (developers) play the music, we just dance. 

Prusa still doesn't have hole correction. But in Prusa and Cura "+" means enlarge (enlarge hole or enlarge outer surface) and that is logical. Or it became logical. In Super Slicer we have something Super 👯 :)

supermerill, 2024-05-27:
> As SuperMeril explain to me once, "minus" means shaving off the perimeter material. Inside or outside. **And yes, this doesn't match the actual descriptive text (mouse hover)**. And yes, some other versions reverse that for I don't know what reason.
> 

Where is the descriptive text problem? can't spot it myself.

> Prusa still doesn't have hole correction. But in Prusa and Cura "+" means enlarge (enlarge hole or enlarge outer surface) and that is logical. Or it became logical. In Super Slicer we have something Super 👯 :)

the '-' -> remove material and '+' -> add material was taken from prusa's xy compensation.

supermerill, 2024-05-27:
@madejackson I fix the direction, there was a problem in a previous version, where the + and - where inverted.
it's the outer or the inner setting that bother you?

CuredPrusa, 2024-05-27:
Hi Super! :)
Here it is:
[image]
It says negative inwards positive outwards.
If you relate that to a hole, it means negative is pushing the hole inwards, meaning the hole will become smaller.  
But we used to do it the opposite, because that's how it works.

madejackson, 2024-05-27:
Yes, i think it's only the Inner one that's reversed.

supermerill, 2024-05-27:
> He means the hover popup text. The settings behaviour doesn't align it's description, it's reversed.

I fail to see how it's reversed. Maybe my wording is bad.
I can change them to something else, maybe 'inward' and 'outward' are confusing.
'inward' here means 'from unprinted area to the printed area'

so for a hole, inward means that the hole will grow as the hole grow into the printed area.

Do you have any idea for a better wording?

madejackson, 2024-05-27:
IMHO the description itself is fine. At least I understand it. Only that it actually does it the other way around in the latest version.

CuredPrusa, 2024-05-27:
>the '-' -> remove area and '+' -> add area was taken from prusa's xy compensation.

Yes, but Prusa doesn't have inner and outer wall separated, that's the catch!

CuredPrusa, 2024-05-27:
The logic from everyday life is simple:
I want to enlarge the hole, I go to hole settings and enlarge it with "plus".
I want to enlarge the outer contour, I go to outer settings and enlarge it with "plus".

But here we have something different - the hole is enlarged with "minus". And that could be fine too as long as we get used to it and it should stays CONSISTANT within all versions otherwise it could get really confusing. And instruction text should be better defined.

madejackson, 2024-05-27:
Yeah I see now how the description can be interpreted both ways...

supermerill, 2024-05-27:
I addad 'add /remove area' to the tooltip. Should be a bit easier now to understand.

> but Prusa doesn't have inner and outer wall separated, that's the catch!

Put the same value on both, to have the same result as in prusaslicer.

>  it should stays CONSISTANT within all versions otherwise it could get really confusing. And instruction text should be better defined.

Agree.

closed 2024-06-12 by supermerill