supermerill/SuperSlicer #4277: M107 called unnecessarily and far too often: Klipper halts (Timer too close)

madejackson, 2024-05-24:
**Bug description**
Superslicer inserts high amounts of M107 / fan stop calls (about 300'000x in my case), even when disabling Fan control completely (0x M106 calls in gcode). It's so much, at one point in my specific gcode, it overloads my MCU and halts the print. The gcode File size is reduced ~10% by just deleting all the useless M107 lines. Both, Arachne & Classic are affected.

[More Info](https://klipper.discourse.group/t/timer-too-close-always-at-the-same-point-in-gcode/16643)

**Expected behavior**
Only generate M107 gcode after M106 was issued previously.

!!>> Project File <<!!
[D0_bearing_based_spirographic_WDT_tool_2024_14.zip](https://github.com/supermerill/SuperSlicer/files/15431013/D0_bearing_based_spirographic_WDT_tool_2024_14.zip)

**To Reproduce**
- Slice .3mf file and print it or check the logs / amount of M107. --> It'll overload MCU at Layer 27 / Z5.5.
- Manually Remove all lines with M107 in gcode. --> Prints perfectly fine.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linuxand Windows
 - Version 2.5.59.9 and 2.5.59.10
 - Printer model: Dueling Zero

**Additional context**
Add any other context about the problem here.

CuredPrusa, 2024-05-24:
I think this could be related to #4225.

madejackson, 2024-05-24:
It may be related but it's not completely the same. My issue startet appearing from 2.5.59.6, 2.4-2.5.59.5 is not affected.
Also The issue appears even when disabling all fan speed control when no M106-Commands are present in the gcode.

supermerill, 2024-05-27:
I'll see if 4225 solve the issue.
also #4168

madejackson, 2024-09-05:
This issue seems to be fixed with version 2.5.60.0, the example gcode now has only 1x M107 occurrence instead of the previous 248'774x.

closed 2024-09-05 by madejackson