supermerill/SuperSlicer #4282: SPEED IS STRANGE.

FoldZero, 2024-05-25:
### What happened?

[image]

inner perimeter is set to 100% in slicer - comes out at 300m/s on print.
MAX speed on slicer is 200.  
so 100% of max speed is 300m/s

Whats going on here?

### Project file & How to reproduce

[speedissues_czfa_flexframe_screw_NoSupport.zip](https://github.com/supermerill/SuperSlicer/files/15444052/speedissues_czfa_flexframe_screw_NoSupport.zip)

### Version

Version: 2.5.59.10

### Operating system

Win10

### Printer model

cr6 MAX

CuredPrusa, 2024-05-25:
Max  speed (which you refer to =200mm/s) is used for auto speed function, when you set 0 at certain speed parameter.
Don't use percentage in Default speed. 
Just put there speed in mm/s at which you want to print. For example 100mm/s. From there you can set perimeter speed for example 200% =200mm/s, and other speeds accordingly.
Check another place of possible speed restriction:
filament_max_speed
filament_max_volumetric_speed

supermerill, 2024-05-27:
as CuredPrusa said, the max speed is a max auto-speed, i'll change the label.

for your speed,  your x maximum feedrate is 300: 
[image]
and your filament max speed isn't defined, so evrything is as expected, I think

FoldZero, 2024-05-30:
is feed rate inclusive of travel speed? I want to have a 250 300 travel speed, but dont neccesarily want to print at those speeds, for quality or flow reasons.

supermerill, 2024-06-03:
it shouldn't

closed 2024-06-12 by supermerill