supermerill/SuperSlicer #4283: New feature: gcode command in Height Range Modifier

CuredPrusa, 2024-05-26:
I think it would be nice to  have ability to add custom gcode command with height range modifier.  With two selectable options:
1. Introduce the custom gcode only at first layer of height range.
2. Introduce the custom gcode at the beginning of each layer within the height range. 

That would improve the manual gcode input opposed to conventional vertical height bar 
- which can not be used for sequential multi printing
- is often difficult to review due to dense layers
- which relates to the layer count, not the height

CuredPrusa, 2024-05-28:
I don't think the option number 2 would be much useful.
Just import custom gcode to the first layer of height range, that should be enough. 
I guess I got trauma from automatically repeated M106 :)

CuredPrusa, 2024-05-28:
>After layer change g-code

Is this the vertical bar for layers?

>{if layer_z >= 0.8 && layer_z < 1};MY COMMAND{endif}

You have to be more specific, I'm not (yet) familiar with coding.

>I'm not sure what the modifier will bring more than that.

Custom gcode in height range modifier would bring better usability of custom gcode, as I already wrote:
- could be used for sequential multi printing
- is easier to review
- relates to the the height , not layer count

supermerill, 2024-05-28:
> Is this the vertical bar for layers?

`After layer change g-code` is here
[image]

> You have to be more specific, I'm not (yet) familiar with coding.

doc: https://help.prusa3d.com/en/article/macros_1775
to be simple:

`{if layer_z >= ` here you enter the start height `&& layer_z < ` here you enter the end height. note that the command will be repeated for each layer taht are between the values. `}`  here you enter your gcode commands. you can add some line return. `{endif}`

. `{` start of the macro
. `if` = if X then command
. ` >=` higher or equal
. `layer_z` the variable that store the current layer height. there is also `layer_num` for the layer number.
. `&&` and
. `}`end of the macro command. As it's a if, the gocde that follow will be printed only if the 'if' is true
.`{endif}` end of the 'if' block.

CuredPrusa, 2024-05-28:
Oh, that! I didn't even pay attention. That could be useful to use. Nice.
Thanks for explanation.
Although this is not as convenient as hrm, but if it's complicated to add gcode input to hrm, this custom G-code in Printer could be a useful workaround. Although, for the intended purpose, it should be located at "Print settings" not "Printer". I usually pull the same "Printer" profile for many projects. 
Thanks anyway.

CuredPrusa, 2024-05-28:
Is it possible to add code to more than one layer with "After layer change g-code "? If so, how to write that?

supermerill, 2024-05-28:
just copy paste and change the value:
```
{if layer_z >= 0.8 && layer_z < 1};MY COMMAND FOR LAYER HEIGH 0.8{endif}
{if layer_z >= 1 && layer_z < 1.2};MY COMMAND FOR LAYER HEIGH 1{endif}
```

CuredPrusa, 2024-06-13:
So will you leave it like this, to use the existing macro before-after layer, or will you add it to HRM? 
I'm fine with both solutions, as long as it works, but it would be more elegant to have it in HRM for better transparency and convenience. I guess a link inside software from HRM to layer macro could be the easiest path.

supermerill, 2024-06-13:
In next nightly, there is object_gcode and region_gcode.
I added them for the pressure advance calibration, but I fear they may be of low usefullness.
But now you have them.