supermerill/SuperSlicer #4284: Very low infill, will not update when changed

Rohrtech, 2024-05-26:
### What happened?

Infill % is extremely low. Does not change when infill % is modified.
[image]
[image]

### Project file & How to reproduce

Slice file with Ubuntu 2.5.59 appimage on Ubuntu 20.04.6 LTS
[Project1.zip](https://github.com/supermerill/SuperSlicer/files/15448256/Project1.zip)

### Version

2.5.59

### Operating system

Ubuntu 20.04.6 LTS

### Printer model

CR-10S

supermerill, 2024-05-27:
probably duplicate of #4243

Rohrtech, 2024-06-03:
Yep, seems to be the case. Sorry for the duplicate! Will wait to upgrade until the next release.