supermerill/SuperSlicer #4286: Allow "Ironing post-process" for print setting with 0 top layers

Phil1988, 2024-05-28:
**Is your feature request related to a problem? Please describe.**
I want to print with aligned infill and then ironing in the same direction for transparent materials.
Aligned infill is possible by setting this:
[image]
[image]

layer 30:
[image]

layer 31:
[image]
So infill works good.

Problem comes when I want to iron in the same direction only.
I neet to set top layers to 0, otherwise it will add "features", see layer 31 now:
[image]
This will destroy transparency to a huge extend.

If I set top layers to 0, I can not activate ironing in the post-process:
[image]

Only option then is to use "Ironing" as infill pattern:
[image]

But then there is no option to modify the infill angle:
[image]

**Describe the solution you'd like**
There are 2 possible solutions that I can think of:
1. Allow Ironing post-process to being activated even if you set 0 top layers
2. Give the option to set the ironing angle in the "Advanced infoll options" with the option for "not alternating ironing"

Then we all can print aligned infill with aligned ironing to create more transparent looking objects :)

supermerill, 2024-05-28:
you can also grow your top layer so it takes the whole layer (anchor solid infill).

> see layer 31 with 1 top layer:

I don't have your project file to see in details.

> 2) Give the option to set the ironing angle in the "Advanced infill options" with the option for "not alternating ironing"

I can. It's not hard.

> 3) Give the option for "aligned" solid infill

This don't makes sense. here "aligned" is in the Z direction. I should make it clearer.

Phil1988, 2024-05-29:
> you can also grow your top layer so it takes the whole layer (anchor solid infill).
> 
> > see layer 31 with 1 top layer:
> 
> I don't have your project file to see in details.
Yeah I thougt the screenshot would be sufficient, but you are right, a project file makes it easier :)

I hope you have an idea how this aligned (in x-y direction) ironing could be achieved in every layer!
Created with SuperSlicer_2.5.59.10_win64_240521:
 [(SuperSlicer)_Transparent.zip](https://github.com/supermerill/SuperSlicer/files/15479381/SuperSlicer._Transparent.zip)

supermerill, 2024-05-29:
[4286_(SuperSlicer)_Transparent.zip](https://github.com/supermerill/SuperSlicer/files/15480262/4286_.SuperSlicer._Transparent.zip)

> Isn't this what I want?

by aligned in z, it means the holes of sparse extrusions are on top of each other. It doesn't do anything when it's a solid infill, as there is no hole.

Phil1988, 2024-05-29:
How do I now make the ironing match the infill alignment?
ironing direction on eg. layer 30:
[image]

should be as eg layer 29:
[image]

So no alternating orientations.
Is this possible?

supermerill, 2024-05-29:
in next nightly:
without 'Alternate Fill Angle', ironing pattern & post-process ironing won't add a 90° angle anymore.

Phil1988, 2024-05-29:
You are a beast!
I will update and close this as soon as I tested it (if you didnt closed it by then already.
Thanks!

closed 2024-06-12 by supermerill

Phil1988, 2024-06-13:
I dont know how to make it work :(
(see attached)
I already tested SuperSlicer_2.5.59.11_win64_240605 before and thought it was not included yet.

Setting1:
[image]
[image]

Setting2:
[image]
[image]

what to activate where? :D 
[settings.zip](https://github.com/user-attachments/files/15812855/settings.zip)