supermerill/SuperSlicer #4287: Sometimes the speed overview includes the travel move

CuredPrusa, 2024-05-28:
I can't define to what it could be related, but sometimes I see that speed overview includes a travel speed and that messes up the whole overview. I can not see nuances of layer speed anymore due to wide spectre of speed range. Sometimes it works fine (without travel speed - as it should be),  but then I change something and travel speed jumps in and I can not get rid of it, so I have to reduce the travel speed to the layer speed to be able to see the speed of layers in colour. 
It happens both in 2.4 and 2.5.
Here's how it looks like:

[image]

Layer speed is 30, travel is 200.

supermerill, 2024-05-28:
you can use the "max" field you have, in the bottom of your screenshot, to eliminate high speed.

I'll look into it on why and how.

CuredPrusa, 2024-05-29:
I don't think this works properly either:

[image]

supermerill, 2024-05-29:
can't reproduce, can you share your project, or how to make the travel included in the graph?

CuredPrusa, 2024-05-29:
I can't either  :)
I wonder if this is a cursed project.
Here it is at your disposal, I gave up on trying to understand this speed overview.....
[test.zip](https://github.com/supermerill/SuperSlicer/files/15481736/test.zip)

While your at it, please check  why other marks are shown on display out of turn, for example this travel around seam (red vertical line):

[image]

And try to see how the print goes at layer 1.60mm. Does this look normal on your computer? Maybe it's my computer, idk.

[image]

supermerill, 2024-05-29:
ok.
If you have selected "travel" as visible, and then you reslice, then they will be included.

CuredPrusa, 2024-05-29:
>If you have selected "travel" as visible, and then you reslice, then they will be included.

Nice! At least I know now how to get rid of travel speed on display. Thanks! I hope you agree travel speed should by all means not be included  in speed overview.

supermerill, 2024-05-29:
In prusaslicer, it's included is shown.
Currently, I replicate the behaviour.

CuredPrusa, 2024-05-29:
Ok.
What do you say about other two strange things?

CuredPrusa, 2024-05-29:
Ok, as long as you're aware of the quirks, I'm fine with that. 
Thanks.

CuredPrusa, 2024-05-31:
>If you have selected "travel" as visible, and then you reslice, then they will be included.

I wish SS can reslice (Ctrl+R) every time, regardless if there's a change or not.

supermerill, 2024-06-03:
it will in next nightly

closed 2024-06-12 by supermerill

CuredPrusa, 2024-06-13:
In 2.5  .12 still can not reslice unless there's a change.