supermerill/SuperSlicer #4288: Add option to relate infill angle to object orientation

TillScout, 2024-05-29:
I am on a CoreXY machine and SuperSlicer is set to add objects at a 45° angle by default, so that for rectangular objects the outside perimeters are done with only a single motor working and seam placement is easier. Consequently, I changed my *fill_angle* to 0, so that the infill is at a 45° angle to the perimeters. However when I have a large object, I have to re-orient it so that it fits my build plate and I often forget to adjust the *fill_angle*.
So my request is if you could relate the *fill_angle* to the object angle, so that with rotation of the object the infill angle will also rotate automatically.

supermerill, 2024-05-29:
that means if I import a file, and rotate it (because the file was in a random orientation, silly designer), then the infill is in this random direction?

CuredPrusa, 2024-06-13:
Isn't that similar to #4304 ?