supermerill/SuperSlicer #4291: Coordinate outside allowed range

AboloneAD, 2024-05-31:
### What happened?

Error "Coordinate outside allowed range" when slicing object with infill more than 0%.
SS version 2.5.59.9 and 2.5.59.10. No problems with 2.5.59.8.
Macos 10.15.7

[image]

### Project file & How to reproduce

Add random object in new project.

### Version

2.5.59.10

### Operating system

Macos 10.15.7

### Printer model

Voron 2.4 SB

cmandlbaur, 2024-06-01:
Getting the same, Arch Linux, 2.5.59.10, Voron 2.2 AB.

cmandlbaur, 2024-06-01:
Installing updated configuration seems to have solved it for me.

cmandlbaur, 2024-06-01:
_seems_ like it's tied to doing infill of 40% or greater.

supermerill, 2024-06-03:
https://github.com/supermerill/SuperSlicer/issues/4257#issuecomment-2135035386