supermerill/SuperSlicer #4293: Retraction Wipe, Wipe only when crossing perimeters - dysfunctional

CuredPrusa, 2024-06-02:
Did I miss something or it's true that checkbox for "Retraction Wipe - Wipe only when crossing perimeters" doesn't work for vast range of SS version, including the latest? 

[image]

I discovered that just recently when I didn't want retraction wipe unless nozzle goes outside the model's wall.

CuredPrusa, 2024-06-02:
Workaround: Retract only when crossing perimeter =true + Wipe while retracting =true + Retract on layer change =false

That brings similar functionality but disables retractions inside the model. 
Nevertheless, this checkbox looks dysfunctional.

supermerill, 2024-06-03:
There was a typo, it needs `avoid_crossing_perimeters` to be able to be enabled.
other than that, it works as expected:
with
[image]
without
[image]

CuredPrusa, 2024-06-03:
Yeah, it's a bit awkward. 
However, I can not get all retractions inside the model and wipe only when crossing perimeter while crossing perimeter is NOT restricted. That's the whole point of wipe - to make crossing perimeter nicer. If I restrict crossing perimeter, I don't need wipe. If I don't restrict crossing perimeter, I actually need wipe. But I don't want it to wipe at every single retraction. I need wipe only when crossing perimeter, while perimeter crossing is not restricted.
[image]
[image]
[image]
This combination of check boxes gives me this:
[image]
But I want this, with free crossing perimeter:
[image]

CuredPrusa, 2024-06-03:
You can test this on project which I have already sent you (test4) with this modifier:
[image]

supermerill, 2024-06-03:
understood, i'll check if the restriction can be lifted without too much work.

supermerill, 2024-06-04:
The issue is that currently, wipe can only be done with a retraction. To decouple the two, a good amount of work & testing is needed.

CuredPrusa, 2024-06-04:
Well, wipe should be done with a retraction. But retraction should not be done with wipe ;)
I could use a workaround (avoid_crossing_perimeters_max_detour =1mm) .
Is it also difficult to disable wipe at last retraction before travel to seam?
(last screenshot here)

supermerill, 2024-06-04:
if it's the last travel, maybe there is a layer change after, and if the retraction is enforced for the layer change, there will be a wipe?

CuredPrusa, 2024-06-08:
Yes, true, retraction is enforced before layer change. But wipe is not necessary if it is nor crossing perimeter. Wipe only when crossing perimeter.