supermerill/SuperSlicer #4294: Infill, Advanced, Only retract when crossing perimeter - threshold too far outwards

CuredPrusa, 2024-06-02:
Version: 2.5.60.0
Build: SuperSlicer_2.5.60.0_2024-05-21
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19045), 64-bit edition

"Infill, Advanced, Only retract when crossing perimeter" - doesn't retract when travel line is on the edge.
It leaves visible lines on otherwise smooth wall surface of the model.
Workaround: I change infill density, travel line gets another exiting point, slicer recognizes "outside perimeter" and retracts.
That works, but after a wasted print which didn't pass quality check. I think slicer should retract before leaving the travel marks on the model surface for being to close to the outer perimeter. 

[image]
[image]

supermerill, 2024-06-03:
can i have your project file?

CuredPrusa, 2024-06-03:
Here it is. 
Problem is at layers 348-371
[test4.zip](https://github.com/user-attachments/files/15535784/test4.zip)

closed 2024-06-12 by supermerill