supermerill/SuperSlicer #4296: M83 not being generated

james56b, 2024-06-02:
### What happened?

When generating a single layer square and with the 'Use Relative E Distances' checked, the slicer does not generate the M83 command in the GCode leading to no extrusion. When manually adding the M83 to the 'Start GCode' field it works without issue.
Toggling the 'Use Relative E distances' off or off/on does not change the situation.

[image]

[image]

### Project file & How to reproduce

[M83 not generated.3mf.zip](https://github.com/user-attachments/files/15524660/M83.not.generated.3mf.zip)

### Version

2.5.59.10

### Operating system

Latest windows 11 & MacOS both

### Printer model

Ratrig VC3 - RatOS RC 2.1

supermerill, 2024-06-03:
as you have `only custom start g-code` enabled, M83 isn't written in the gocde file, as these may be written in a firmware macro, or somewhere else.
In your custom start g-code, you may want to add:
```
G21 ; set units to millimeters
G90 ; use absolute coordinates
{if use_relative_e_distances}M83{else}M82{endif}
G92 E0 ; reset extrusion distance
```

james56b, 2024-06-06:
Thanks Merill. That was indeed a solution. 
With some help an alternative was also to set the 'Gcode Flavor' to 'Klipper'. This generated the G21,G90, and M83 as well.

closed 2024-06-06 by james56b

Pertinal, 2024-06-07:
> Thanks Merill. That was indeed a solution. With some help an alternative was also to set the 'Gcode Flavor' to 'Klipper'. This generated the G21,G90, and M83 as well.

Small correction - it did not. Have had to manually add the m83, g21, and g90 to the start gcode for ratrig printers.