supermerill/SuperSlicer #4298: SuperSlicer 2.5.59.10 Windows 11 Crash

robertspark, 2024-06-03:
### What happened?

Superslicer shuts down unexpectedly and without warning / message

### Project file & How to reproduce

The issue does not appear to be project specific (I've done the same thing with several projects).   
IT seems to happen without warning / at a non specific time.   IT crashed when I was rotating an object, and it has also crashed after creating gcode, and it has also crashed after I've saved a file.   IT has just crashed now after opening superslicer and having it running a short while.

[superslicer crash.csv](https://github.com/user-attachments/files/15534028/superslicer.crash.csv)

Attached are the windows logs in csv file format.

And within the zipper folder is a config bundle ini, project file and gcode export for the project file.
[SuperSlicer_config_bundle.zip](https://github.com/user-attachments/files/15534111/SuperSlicer_config_bundle.zip)

The issue does not seem project specific though and does seem to crash after a timeout period so it happens randomly, regardless of what is being done at the time.

### Version

2.5.59.10 

### Operating system

windows 11 home (10.0.22631 build 22631)

### Printer model

custom klipper setup with a highly modified CR10S

supermerill, 2024-06-03:
Here is a debug build: https://drive.google.com/file/d/1spJ-zkQgT2HH4qg3AosGQDbH3AJ3Od8n/view?usp=sharing
it's very slow to slice, but if it crash, the crash log will be useful.
You can also try this release build: https://github.com/supermerill/SuperSlicer/actions/runs/9265610061/artifacts/1543286106 maybe the bigfix it contains resolve the issue?

roykrikke, 2024-06-09:
Is this more or less the same as https://github.com/supermerill/SuperSlicer/issues/4301?

robertspark, 2024-06-09:
no, it crashes even without slicing, I just open the program, give it a few minutes, not do anything and it will crash.

or I can open something (anything) slice it and save it and providing I am quick it will save the sliced gcode for upload ..... then it crashes....

I tried the alternative logging version provided and I could not even get it to load (tried running it as an administrator too... will not even load or start)... gave up been busy on other stuff 

will try again later, please note this is W11 home, (all my other pcs and laptops are W10 or Linux.... I've not tried this version on any of them yet (I will try to give them a shot with the same version on w10 and Linux (Debian))

robertspark, 2024-06-09:
I did wonder if it was related to adding the IP address for my klipper printer into superslicer for direct upload....  I wanted to try removing this and see if that sorts the issue