supermerill/SuperSlicer #4300: Floating Brim (Sequential Printing, Supports)

Dobarekk, 2024-06-04:
### What happened?

If I slice my models with sequential printing enabled and using "Supports Everywhere", it produces 2 Brims.
One on Layer 1 as expected and one on Layer 8.
This only happens when both sequential printing and Supports everywhere is enabled.
If either of them aren't (even only support on buildplate), this doesn't happen.

### Project file & How to reproduce

- Screenshot from Bottom for better visual distinction
- Zip includes obj, mtl, 3mf files (don't know if this is the right way to export)
[image]
[a.zip](https://github.com/user-attachments/files/15571238/a.zip)

### Version

2.5.59.9

### Operating system

Windows 11

### Printer model

Voron 2.4 with Afterburner (Mostly default Profile)

supermerill, 2024-06-05:
It tried to put the skirt also on the first support layer, and when this one isn't on the bed... double skirt.
fixed.
In the meantime, you may have to use the "allow only one skirt loop" `complete_objects_one_skirt`

Dobarekk, 2024-06-10:
Thank you very much for both the fix and the workaround.
People like you is what makes this community is so great.
I love your work, keep it up <3

closed 2024-07-01 by supermerill