supermerill/SuperSlicer #4301: Crashing and closing when slicing

GalvanicGlaze, 2024-06-05:
### What happened?

When selecting Gcode viewer and while slicing my file it crashes and exits

### Project file & How to reproduce

[red.zip](https://github.com/user-attachments/files/15587636/red.zip)

### Version

2.5.59.11

### Operating system

Windows 10

### Printer model

Voron Trident

fbeauKmi, 2024-06-06:
Same issue with this project
[crash_2.59.11.3mf.zip](https://github.com/user-attachments/files/15614114/crash_2.59.11.3mf.zip)
no crash log

Crash occures generating gcode for layer 59 (last layer of the smallest object)
[image]

### Version:
from 2.5.59.11 to nightly-dev (2.5.60.0)

### Operating system:
Windows 11

Tehvan, 2024-06-07:
Same issue. 2.5.59.10 slices but 2.5.59.11 won't. This is the crash log.
[crash_log.txt](https://github.com/user-attachments/files/15710616/crash_log.txt)
And here are the files I was trying to slice.
[Shuttle_MGN12H.zip](https://github.com/user-attachments/files/15710623/Shuttle_MGN12H.zip)

roykrikke, 2024-06-08:
I have the same issue with 2.5.59.11 on windows 10.

I'm willing to share a crash.log.txt But don't know where to get the file.

And is the files I was trying to slice. 
[LDO_LED_PCB_spacer_narrow_rev01.zip](https://github.com/user-attachments/files/15747660/LDO_LED_PCB_spacer_narrow_rev01.zip)

OldCurmudgeon3DP, 2024-06-08:
Similar issue, but crashes/closes while slicing if I select "No" when the "Mutli-part object detected" dialog appears.
2.5.59.11 and alpha 728 for windows.

roykrikke, 2024-06-09:
> I have the same issue with 2.5.59.11 on windows 10. With version 2.5.59.9 slicing is working without the crash.
> 
> I'm willing to share a crash.log.txt But don't know where to get the file.
> 
> And is the files I was trying to slice. Crash @ layer 21. [LDO_LED_PCB_spacer_narrow_rev01.zip](https://github.com/user-attachments/files/15747660/LDO_LED_PCB_spacer_narrow_rev01.zip)

I did some more testing with regardig this issue with 2.5.59.11 on windows 10. 

If I merge all the parts in my shared file, there is no crash. I wonder if others can test this also and share there results.

fbeauKmi, 2024-06-09:
> 
> I did some more testing with regardig this issue with 2.5.59.11 on windows 10.
> 
> If I merge all the parts in my shared file, there is no crash. I wonder if others can test this also and share there results.
Done with my project , it doesn't crash when merged too, thanks.

jaminmc, 2024-06-11:
Downloaded the nightly build from the actions section, and it fixed my problem of slicing. If I had more than one .stl, it would crash when slicing, Now it works great! :)

I am using MacOS, arm version.

closed 2024-07-01 by supermerill