supermerill/SuperSlicer #4302: Error key:2000

StrDa121, 2024-06-06:
Hi,
sometimes i have problem with printing small objects such as a gear.
Printing stops with a warning key:2000.
The problem occurs when the layer height is set to 0.1 mm.

I use SuperSlicer v.2.5.59.11
Windows 10
Creality Ender 3 V3 KE

Thanks
[Gear.zip](https://github.com/user-attachments/files/15673193/Gear.zip)

supermerill, 2024-06-10:
How often it is?
it's a warning after the gcode has been generated or before clicking on the slice button?
Can you take a screen shot of the warning?

thanks.

justvlade, 2024-06-21:
He says that his printer shows this message. I do not own such printer, but I googled a bit. Looks like it is hardware/firmware problem (well obviously printers should not break from gcode). Some people change boards, some people try to upgrade firmware. Some say Orca slicer gcode does not trigger this error.
I did not google to much, but I have an opinion no one exactly knows how to fix this error or what causes it.
But it is not a slicing problem. I guess.

Izzatzainihage, 2024-08-10:
Got the same issue, error 2000  while printing. i try everyting to solve it, but it solved after using orca slicer,