supermerill/SuperSlicer #4304: Infill pattern doesn't rotate with the object

CuredPrusa, 2024-06-08:
Related to all SS versions:
Infill has a defined filling angle. 
When instances multiply and rotate around the build plate, the infill angle doesn't follow the model position/rotation, but a relative position to the build plate. That messes up complete infill pattern, otherwise precisely defined for object's  requirements.  

[image]

confuzled-dev, 2024-06-08:
been frustrated by this for a while

CuredPrusa, 2024-06-08:
I'm actually quite amazed this wasn't fixed years ago. I have found this same flaw in Cura and Orca too. Very peculiar phenomenon.

supermerill, 2024-06-10:
I can adjust it but when you have multiple part into an object, the rotation should be:
 * the plater (current)
 * only the rotation of the object
 * only the rotation of the part
 * the combined rotation of the object + part
?

CuredPrusa, 2024-06-11:
I can't imagine when and why should infill angle be related to the platter. Whenever the infill angle matters, it's always adjusted to the appropriate fit inside particular perimeter. So any consequential automatic change of the infill angle in relation to the perimeter, for whatever reason (in this case rotation, regardless if it's multi or solo), is actually a problem which has to be manually inspected and manually corrected, to preserve infill angle related to perimeter as originally desired. 

So to put it pragmatic, the infill connection points to the perimeter should be considered as GLUED  to the perimeter in original object position, for whatever rotation it done. I guess this means that  angle degrees should be calculated to perimeter plan. Yes, there could be many perimeters in various positions, rounded as well, but starting position doesn't matter much because user adjust afterwards precisely when observing, inspecting. Or the starting point could be related to platter, but once the infill is there it should become a matter of perimeter. The important thing is it should stay the way it was set.

supermerill, 2024-06-11:
Here I set manually the first object to 0° angle and the second to 45°, as it's rotated by 45°, is it what you want the software to do automatically?
[image]

confuzled-dev, 2024-06-11:
Maybe an easy right click "set infill angle" would be better. You are never
going to have the infill right all the time based on a preconfigured
setting. So just make it easier to modify

CuredPrusa, 2024-06-12:
>the infill connection points to the perimeter should be considered as GLUED to the perimeter in original

>imo, not possible if the object is rotated/cut/modified/assemble with others.

That's a pity because this is exactly what I have in mind. 
If this is not possible, that means duplicating object by instances and rotate them around the platter for better convenience of printing is actually impossible because infill pattern would always shift. Example is in screenshot of my first post. To correct that behaviour, I'd have to copy-paste complete object  and manually change infill angle and solid fill and ironing for each copy of original object through bunch of individual modifiers. Yes, it's a doable workaround, but it's tedious and frustrating, especially for complex objects with several infill chambers and manually arranged infill patterns and density. 

> You are never going to have the infill right all the time based on a preconfigured setting. 

I never thought about it. Now, this changes the whole approach in rearranging the position of the object(s).

CuredPrusa, 2024-06-13:
As I already made it clear, I'm not familiar with coding of such a complex program as slicer. But, what comes to my mind is to make a check box which adds or subtracts angle degrees to anything related to filling/ironing, in direct relation to XYZ rotation of the object.

supermerill, 2024-06-13:
added a setting to follow the object's rotation, as it's easy enough to add and it makes sense.
For the cases where the rotation is too strange, the fill_angle field is added to the quick-modifier 'infill'