supermerill/SuperSlicer #4308: Crash when trying to slice a model

Skorki51, 2024-06-09:
### Version
2.5.59.11

### Operating system type + version
Windows 11 Pro / 10.0.22631 22631

### Behavior
* Crash when trying to slice a model

* When I tried to change the values ​​in the XY compensation, the program crashed.
Everything works fine with these settings:
[image]

* But when I change the external or internal compensation, the program crashes.
e.g. for such values:
[image]

#### Project File (.3MF) where problem occurs
[project.zip](https://github.com/user-attachments/files/15750795/project.zip)

roykrikke, 2024-06-09:
Is this the same as https://github.com/supermerill/SuperSlicer/issues/4301?

Skorki51, 2024-06-09:
Possibly because it looks similar. 

But it doesn't always crash. depends on XY compensation settings. With this model that I added -0.05 and -0.1, the program works properly, but with other values ​​it crashes.

With another model, the values ​​-0.05 and -0.01 also did not cause a problem, -0.05 and -0.2 also worked, but something between, e.g. -0.05 and -0.015 crashed.

closed 2024-07-01 by supermerill