supermerill/SuperSlicer #4309: Crash when Slicing

Tirpitz93, 2024-06-09:
### What happened?

Upgraded to 2.59.11 and now it crashes every time I try to slice a model.
Slices fine in 2.59.10

### Project file & How to reproduce

[switch_programming_plate_pm2.zip](https://github.com/user-attachments/files/15751430/switch_programming_plate_pm2.zip)

### Version

2.5.59.11

### Operating system

Windows 11

### Printer model

Prusa MK2.5s MMU2S

Tirpitz93, 2024-06-09:
[CrashDumps.zip](https://github.com/user-attachments/files/15751463/CrashDumps.zip)

justvlade, 2024-06-10:
I confirm this project "switch_programming_plate_pm2.3mf" also cause SuperSlicer-ubuntu_20.04-2.5.59.11.AppImage crash, under Linux Mint 21 (Which is Ubuntu 20.04 as far as I understand). And under version 2.5.59.10 it slicers OK.

BUT = )
I also have a project, which crashes in 2.5.59.10 and in 2.5.59.11. While in 2.5.59.10 and 2.5.59.09 it says "std::bad_alloc" and keeps open, on 2.5.59.11 it says nothing and just closes the app. The version 2.5.59.8 slices it without error, while yelling about unrecognized parameters while opening and loading project. I have checked those under Mint 21. Here is my project: 
[111.3mf.zip](https://github.com/user-attachments/files/15780175/111.3mf.zip)
Also checked 2.5.59.11 version under VirtualBox Ubuntu 24.04 it also crashes silently. I created this project there with default Ender 3 0.4 printer and default profiles which it automatically activated.

Tirpitz93, 2024-06-17:
I will test tonight, Thank you

closed 2024-07-01 by supermerill