supermerill/SuperSlicer #4311: SuperSlicer MacOS ARM Release Builds Are Corrupt

jsaintrocc, 2024-06-09:
### What happened?

I have a MacBook Pro with an M1 Pro chip so I downloaded a MacOS Arm release from [the release page](https://github.com/supermerill/SuperSlicer/releases). When I mount the image and try and run the SuperSlicer binary I get the attached error. I duplicated this for all releases that have an arm build. I also tried copying the binary to my application directory and received the same error. The non-arm MacOS builds work as expected.

[image]

Here is a screenshot of my hardware and OS
[image]

### Project file & How to reproduce

## Procedure

1. Download [SuperSlicer_2.5.59.11_macos_arm_240605.dmg](https://github.com/supermerill/SuperSlicer/releases/download/2.5.59.11/SuperSlicer_2.5.59.11_macos_arm_240605.dmg)
2. Doubleclick the dmg to mount the image
3. Doubleclick the executable

## Expected Result

SuperSlicer launches

## Actual Result

Receive the error message `"SuperSlicer" is damaged and can't be opened. You should eject the disk image.`

### Version

2.5.59.11

### Operating system

macOS Sonoma Version 14.5

### Printer model

voron 2.4r2

Pieterv24, 2024-06-11:
I am experiencing the same issues on a M3 Macbook.

When running the non arm build, I have no issues however.

supermerill, 2024-06-11:
duplicate of #4253 (and #4239)
try with xattr , as I didn't sign the dmg (yet)

jaminmc, 2024-06-11:
> duplicate of #4253 (and #4239) try with xattr , as I didn't sign the dmg (yet)

Works fine with a `xattr -rd com.apple.quarantine /Applications/SuperSlicer.app` -- assuming you put it in your applications folder.

jsaintrocc, 2024-06-21:
Thanks @jaminmc. Skipping the malware check did it and I  can now launch the ARM version. I see it's now in the release notes which I was going to suggest. Cool!

commit 6e283f69 (2018-07-30): Flush the output buffer when writing svg files from PrintSLA.cpp.

Fixes #4311
closed 2024-06-21 by jsaintrocc