supermerill/SuperSlicer #4312: Error while writing gcode: two points are at the same position

treeman, 2024-06-10:
### What happened?

Got the error

```
Error while writing gcode: two points are at the same position
```

When slicing the model.

### Project file & How to reproduce

[v3-case-right.zip](https://github.com/user-attachments/files/15756886/v3-case-right.zip)

### Version

2.5.59.11

### Operating system

Void Linux

### Printer model

Voron Trident

supermerill, 2024-06-11:
can't reproduce on windows, will try on ubuntu.

supermerill, 2024-06-11:
same, can't reproduce.
Can you give me your project file instead of just the stl?

treeman, 2024-06-11:
[cosmo.zip](https://github.com/user-attachments/files/15792163/cosmo.zip)

Sorry, my bad.

treeman, 2024-06-15:
I updated to the new version [2.5.59.12](https://github.com/supermerill/SuperSlicer/releases/tag/2.5.59.12) and the problem has disappeared.

closed 2024-06-15 by treeman