supermerill/SuperSlicer #4318: Merging a merged object with another object changes the rotation state of a part of the original object

Icenowy, 2024-06-13:
### What happened?

I am trying to hack a customized Voron skirt in SuperSlicer when discovering this problem. When merging the Merged object with another simple object, some parts of the Merged object changes rotation.

### Project file & How to reproduce

[Merged.zip](https://github.com/user-attachments/files/15814763/Merged.zip)

Merge the `Merged` object with any `Shape-Box` object and the problem will happen.

### Version

2.5.59.12

### Operating system

AOSC OS (distro built SuperSlicer)

supermerill, 2024-06-13:
mmmm
can't reproduce.
Maybe when the project is saved, the rotation is set in a better way?
is it the rotation of the cube that is messed up?

Icenowy, 2024-06-13:
Can't reproduce with 3MF either.

Never mind now.

closed 2024-06-13 by Icenowy