supermerill/SuperSlicer #4320: Add the ability to split a g-code file from inside the g-code viewer application.

Nbeknel, 2024-06-13:
**Is your feature request related to a problem? Please describe.**
There are several solutions on how to save a print that failed midway. One is to reslice the object that is submerged into the bed by the height already printed, another is to manualy modify the existing g-code file by removing the g-code that printed perfectly and adding G92 codes after the start g-code. In both scenarios one will probably open the g-code file in the g-code viewer to verify at which height/layer the print failed.

**Describe the solution you'd like**
I would like the g-code viewer to have the ability to split the existing g-code file and create a new one from the height/layer chosen by the user.

**Describe how it would work**
The function would work like a post-processing script modifying the existing file. In an ideal situation the following should be changed according to the config found at the end of the g-code file:
- "first layer" should be slowed down
- fan speeds should be disabled/changed according to their respective setting
- a brim should be added if present
- raft_first_layer should be added to supports
- skirt added around the new "first layer"

**Describe alternatives you've considered**
I am considering writing my own post-processing script, but currently am unable to find the time, in addition, this would be yet another script, separated from the SuperSlicer. I'd like for it to be an integrated process.

Nbeknel, 2024-06-17:
From the bed again. One might not always be beside the printer when the print fails, while the printer continues "printing" and eventually finishes. When that happens, the printer may disengage the motors and disable the bed heater. If we try to print on top of what's already printed, we run into the problems of the part hindering the homing process, or the part disengaging from the print surface as it'll have cooled down.

Tirpitz93, 2024-06-17:
I have had luck with the following procedure (Prusa mk2.5s):

0.  Print failed and printer continued and "finished"
1. modify gcode file to remove succesful code, keeping start code 
2. change the Autohome XYZ for a home only XY
3. use calipers (or the print itself) and manual commands/printer interface to get to correct height.
4. Start the new print file.

supermerill, 2024-06-27:
I guess It's possible to create a little tool that do that.
Like an entry in the menu show up a dialog that ask for the height, and what to keep from the start gcode (home xy, home xyz, no home, whole custom gcode)
then export a file with "repair_XXmm" as a suffix