supermerill/SuperSlicer #4322: Internal bridge infill AND Solid infill not touching perimeter due to above layers

Arthur-de-Partuur, 2024-06-14:
### What happened?

When I slice a (particular) part, the internal bridge infill does not reach far enough to the perimeter (on one side only). as such the infill "drops" down and leaves gaps in the surfaces above. Reason that it is not touching the perimeter is (probably) that there is a wall above this layer. As there is only sparse infill below, the infill layers have nothing to connect to or "lay on top of"

### Project file & How to reproduce

[image]
[cradle.zip](https://github.com/user-attachments/files/15837333/cradle.zip)
Slice with 2 perimeters (0.6mm), 3 solid bottom layers and 25% infill. Check layer 8 (and 9)

### Version

2.5.59.11 and 2.5.59.12

### Operating system

Win 11 pro

### Printer model

RatRig V-Core 3.1 running RatOS 2.1 (Klipper)

supermerill, 2024-07-01:
increase the external_infill_margin a bit, like to 250%.
you can also active the supporting dense layer.

Arthur-de-Partuur, 2024-07-01:
Thanx Merill, That 250% is exactly enough,... funny. Not a bug. Issue can be closed

closed 2024-07-01 by Arthur-de-Partuur