supermerill/SuperSlicer #4330: Thin walls bug/issue - inconsistent slicing and thin walls width too small

pajtaz, 2024-06-23:
**Bug description**
There are two issues here, not sure how they are related.
1. Main issue is that thin walls are too small. If using thin walls and having any settings other than 100% under "Min width:" the thin walls end up being extruded at widths that are smaller than that setting and really small. In the attached 3MF slice the file and go to layer 20. You will see thin walls having widths of 0.017934, 0.0265, 0.007767, 0.040366, etc. See attached photo of how that layer external wall looks like in gcode preview - be careful not to look at the end of the layer because it does not look like that during the preview. When printed that layer is essentially missing the external perimeter. I did not take a photo of the printed part but the part is missing that layer 20 external perimeter.
2. Second issue or bug is that the same part oriented 90 degrees does not slice the same way, at least on layer 20. It still ends up with a few really thin walls but the main part of the wall ends up extruded as a normal external perimeter with width of 0.48, while on the original part that same main part of the wall is being extruded as thin wall of width 0.017919.
**Expected behavior**
1. Thin walls should not have line widths smaller than the setting under "Min width:" - if I am understanding that setting correctly.
2. Same part should have exact same line widths no matter which orientation it is placed.

!!>> Project File <<!!
Project file attached; gcode of original orientation attached; gcode of 90 deg rotated attached.
**To Reproduce**
Steps to reproduce the behavior:
1. For issue 1, slice the file, go to gcode and look at layer 20. Go to gcode preview and slide through layer 20 from beginning.
2. For issue 2, orient the model by rotating it 90 deg around Z axis. Slice the file, go to gcode and look at layer 20. Go to gcode preview and slide through layer 20 from beginning.
**Screenshots**
Screenshots of two orientations attached. It is gcode preview at the end of the external perimeter. Screenshot of thin walls setting attached.
[image]
[image]
[image]
[gcodes.zip](https://github.com/user-attachments/files/15945231/gcodes.zip)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.5.59.9 and 2.5.59.12
 - Voron Trident 300mm
[thin_walls_issue.zip](https://github.com/user-attachments/files/15945227/thin_walls_issue.zip)

supermerill, 2024-06-30:
2. yes, there is a bug, fixing it.

1 & 3. I don' t understand, there is only a thin wall at layer 100 & 101. There is only gapfill at layer 20 that is much lower than the perimeter width.

neophyl, 2024-06-30:
Merill the OP seems to be counting layers from the top down rather than referencing the actual layer number.  I couldn't figure out the issue till I looked at layer 100 and matched it to their screen caps. 

There are some VERY thin start/end sections to some of the segments on layer 100.  Looking at the preview I can see where it might give gaps in the perimeters when printed.

supermerill, 2024-06-30:
the taper at the ends of the thin wall?
it's intended, as @VanessaE has tested thin walls a lot, and these taper improve the print quality.
You can disable them by setting the 'thin wall overlap' to 0.

pajtaz, 2024-06-30:
> 2. yes, there is a bug, fixing it. Workaround: uncheck 'merge with perimeters'
> 
> 
> 1 & 3. I don' t understand, there is only a thin wall at layer 100 & 101. There is only gapfill at layer 20 that is much lower than the perimeter width.

I apologize, I always confuse height and layer number. I meant layer 100 (height of 20 mm).

pajtaz, 2024-06-30:
> 2. yes, there is a bug, fixing it. Workaround: uncheck 'merge with perimeters'
> 
> 
> 1 & 3. I don' t understand, there is only a thin wall at layer 100 & 101. There is only gapfill at layer 20 that is much lower than the perimeter width.

For some reason github comment field has made a mess of the numbering. There should not be 3 and 4, only 1 and 2 and "Expected Behavior" 1 and 2. When I click to edit the original comment, it clearly shows 1 and 2 under "Expected Behavior" but after you click save it changes it to 3 and 4.

So issue 1 is the big one. Thin walls are extruded with line widths much smaller than specified amount but maybe I am not understanding the setting.

pajtaz, 2024-06-30:
Oh I fixed the numbering by adding one clear line before "Expected Behavior".

pajtaz, 2024-06-30:
It seems that for this specific model unchecking "Merge with perimeters" kind of fixes the issue. However there are still thin walls extruded at very small widths, which I do not know is the intended behavior (width of 0.07761, for example).

supermerill, 2024-07-01:
> However there are still thin walls extruded at very small widths, which I do not know is the intended behavior (width of 0.07761, for example).

the end and start of thinwalls are tapered, to improve the printing. If your printer has an extremely reliable pressure control, you can reduce it and even eliminating it by changing the 'thin wall overlap' setting.