supermerill/SuperSlicer #4331: Negative space scaling/rotating

sprior, 2024-06-24:
### What happened?

I started with a stl and I wanted to drill a couple of holes in the side.  I scaled the original object, then selected it and created a negative space cylinder which I then scaled to 4.9mm in x and y and 20mm in Z.  Everything's fine so far.  But when I selected that negative space and rotated it 90 degrees around x it got distorted, the end face turned oval and the negative space was displayed more than twice as long.  I did exactly this same exercise in Bambu Studio and it worked fine.

### Project file & How to reproduce

[bug.zip](https://github.com/user-attachments/files/15948503/bug.zip)

### Version

2.5.59.2

### Operating system

Windows 11

### Printer model

voron 2.4 with Stealthburner

supermerill, 2024-06-30:
same in prusa 2.5, they corrected it in 2.7, will be after the merge.

supermerill, 2024-06-30:
a workaround:
add a negative object before doing any rescaling.
don't rescale the main object node, but only the selina.stl part node.
then it should be okay.