supermerill/SuperSlicer #4332: Prime all printing extruders does not take delta radius into consideration

th0mpy, 2024-06-27:
### What happened?

If you select the option to "Prime all printing extruders" and you're running a delta printer, the system generates an error stating that "A toolpathouside the print area was detected" 

I would either expect the wipe area to be pulled into the center further as to not exceed the printer's available radius -OR- allow the user to define the location -OR- make a curved wipe based on the radius

### Project file & How to reproduce

Print settings -> Multiple Extruders -> Prime all printing extruders (check) 
[image]

Slice the multi-color model (error now appears in the sliced view.

[image]

### Version

2.5.59.12

### Operating system

MacOS 14.5

### Printer model

Rostock with bowden splitter and smart-effector

supermerill, 2024-06-27:
I'll add a simple position entry, like for the wipe tower.
Ideally, I'll add a shape to move in the platter, but this will be in a second time.

supermerill, 2024-06-27:
I can't reproduce your problem, can you send here your project file?

th0mpy, 2024-06-27:
Attached is the project file. 
[Calibration cube - rostock error.3mf.zip](https://github.com/user-attachments/files/16014045/Calibration.cube.-.rostock.error.3mf.zip)

Interesting outcome though. I didn't touch the program after I submitted the ticket, the error was present on the screen when I just came down. I saved the project file to upload, closed out and reopened SuperSlicer. I reopened that project, sliced it, and see the tower has now moved. Presumably the error is gone now?