supermerill/SuperSlicer #4335: Slice direction. (PrusaSlicer has in beta/dev)

Tau512, 2024-06-30:
I've been using a revo hotend for a while and noticed that i'd sometimes find the nozzle slightly loose, which contributes to poor layer stacking. I've not put much thought into why and figured i just wasn't tightening the nozzle enough (first attempt at tightening more: so far, so good) as it'd occur almost randomly over a number of prints.

I stumbled upon a possible reason (and fix) in a youtube stream: https://www.youtube.com/live/AOFvM87RdHc?si=eME1-AWJkVFo7m6G&t=8123 (direct link to timepoint (2h15m23sec)).... Change the direction of slicing from CCW to CW.
While i can find a setting specific to overhang behaviour, I cant find a general setting for this in SS.

I know your working on a PS re-basing, so wondered if this would be possible to include, or add the feature in an upcoming release.

supermerill, 2024-06-30:
I don't  really understand. prusaslicer has the feature? if yes, since what version (so i can know if it will be included quickly or i need to cherry-pick it)

Tau512, 2024-06-30:
according to that reddit link i posted, its available in the 2.8.0 PS release.

quote from post:
> And we solved it! Let me introduce you to what I personally call "Revo mode" but is officially "Prefer clockwise movements" in the new [@PrusaSlicer](https://x.com/PrusaSlicer) 2.8.0 (Printer -> General -> Advanced section)

i've done a feature name search in their git, hopefully it's useful to track down the feature:
https://github.com/search?q=repo%3Aprusa3d%2FPrusaSlicer%20Prefer%20clockwise%20movements&type=code

hope that's clearer!