supermerill/SuperSlicer #4336: A ratio for print speed dependent upon overhang angle

Shefoofy, 2024-06-30:
**Is your feature request related to a problem? Please describe.**
SS does not provide a way to slow print speed depending on the angle of the overhang.

**Describe the solution you'd like**
I would like a reduction in print speed automatically as the angle increases

**Describe how it would work**
You would have a ratio of print speed to angle option where you could change the percentage or rate of printing reduced depending on the angle. This could be a percentage or a sliding scale. There could be several ways to do this.

**Describe alternatives you've considered**
Right now, I have to manually change the speed when an overhang is more than 45 degrees and then speed it up when it becomes more vertical

**Additional context**
None that I can think of.

Sirage48, 2024-07-01:
I think that is already in the pipeline when the merge with 2.7 happens. It is a big job so be patient. For software it is finished when it is finished.

Shefoofy, 2024-07-01:
That's great!