supermerill/SuperSlicer #4337: V-Core 4 Profiles

nameisCruzCruz, 2024-07-01:
Hi! I'd like to submit the V-Core 4 printer settings as well as the print settings, along with the PunkFil filament line by RatRig.
I'm sorry if this is not the most convenient spot to supply the files, if this way doesn't work, let me know and I will correct.
[SuperSlicer.zip](https://github.com/user-attachments/files/16053558/SuperSlicer.zip)
Thank you!

supermerill, 2024-07-02:
np
profiles are [here](https://github.com/slic3r/slic3r-profiles) fir submission.
I'll check your profile.

nameisCruzCruz, 2024-07-09:
Hi! I don't mean to put pressure on this subject, I imagine you have a lot of stuff to do. May I ask when do you expect to make the profiles available? Thanks for your time!