supermerill/SuperSlicer #4342: Layer Height to be allowed to Exceed Nozzle Width.

julianjc84, 2024-07-03:
**Feature Request.**
Layer Height to be allowed to Exceed Nozzle Width.
example:
LayerHeight 1mm
[image]
Nozzle Diameter Installed on hotend 0.6mm
[image]

It would be nice to be not restricted by Layer Height and Nozzle Width.
There seems to be no limit of the maximum line width allowed to extrude.
[image]

A 0.6Nozzle can extrude a 2mm width line no problem. Especially with high flow hot ends. Yes there are issues with the "spacing", and manual adjustment of the linewidth spacing in relation to the line width is another consideration issue to consider.

**Describe the solution you'd like**
De-Restrict the relationship between Nozzle Diameter and Max LayerHeight / Allow override.
[image]

**Reason For Request**
Aesthetics look for prints.
Big Layer height with small high flow hot ends.
General functional prints where you want alot of plastic
a 0.6Nozzle can really extrude alot of filament.
[image]

supermerill, 2024-07-03:
there is multi-extruder printer.
You can print an object with a huge extruder and huge layer, and another one with small extruder and small layer height.
That's why the min/max layer height on the extruder tab is checked (that can be changed from slicing to edit) and then each layer is checked against all theirs used extruders min/max lh.

julianjc84, 2024-07-03:
The Nozzle Diameter overrides the layer height limits.
Regardless of Layer height limits when they exceed nozzle diameter. EG: 200% or 10mm
[image]
if the Layer Height is greater than nozzle diameter then it wont slice.
[image]

supermerill, 2024-07-03:
The min/max layer height on the extruder tab is checked on slincing, so it doesn't go higher than the nozzle size.
I can change it so it's checked when someone edit the field, like when you try to set a brim separation gap higher than the brim width.

What i'm try to say is: don't put 200% here, this is an invalid value, as any higher than 100%.

julianjc84, 2024-07-03:
i don't really understand you sorry. but here is gif example

[image]

supermerill, 2024-07-03:
I'll start again from the beginning.

> It would be nice to be not restricted by Layer Height and Nozzle Width.

There is checks to ensure the shape of the extrusion are feasible.

> A 0.6Nozzle can extrude a 2mm width line no problem. Especially with high flow hot ends. 

then change the nozzle diameter from 0.6 to 2, and everything should work.

>Yes there are issues with the "spacing", and manual adjustment of the linewidth spacing in relation to the line width is another consideration issue to consider.

if it's in % and you change the nozzle diameter, then everything should have changed accordingly.

julianjc84, 2024-07-03:
I will try some tests.
I believe I understand it now and it should work as expected.

Thankyou.

closed 2024-07-03 by julianjc84