supermerill/SuperSlicer #4343: Gcode generation problem - mesh size - suspicious result

SimonSolar2C, 2024-07-03:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.5.60.00

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### Behavior
* _Describe the problem_
* Gcode mesh dimensions incorrect according to Klipper
* These are they: Please see screenshot
* _Steps needed to reproduce the problem_
* Generate a file? I can PM file
* _Expected Results_
* Smooth walled print ( or my extruder is blocked - checking )
* 
* _Actual Results_
  * _Screenshots if useful_
  * 
[image]
[image]
[image]

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/
No
#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

SimonSolar2C, 2024-07-03:
Also please see my Gcode macro variables  which worked in 2.5.59
[image]

supermerill, 2024-07-03:
I tried with one of my project, and ican't reproduce. can you save & give me a project that shows the problem? (you dag & drop a zipped .3mf here)

btw, you can use min & max in macro:
`MESH_CONFIG X0={max(0, first_layer_print_min[0]-10)} Y0={max(0, first_layer_print_min[1]-10)} X1={min(350, first_layer_print_max[0]+10)} Y1={min(350, first_layer_print_max[1]+10)} `

SimonSolar2C, 2024-07-03:
Hi Thanks for looking:

 

Attached.  Note this file gave ‘crash-like’ troubles with previous version. Sorry cant reproduce because I installed by overwriting into previous SuperSlicer folder.

Did you change the variable names for the print extents?

 

 

 

Kind regards,
Simon Dyer,
2C Light Company Ltd.
ph +643 981 8376
www.2CLight.com

SimonSolar2C, 2024-07-05:
Just adding a further observation - it seems it has generated the output with the outer perimeter having a very low flow rate, whereas it should be similar to the inner perimeter.  See pics below: 
[image]
[image]
[image]
[image]

supermerill, 2024-07-05:
you can't attach files by email, you need to drag & drop them here manually.

SimonSolar2C, 2024-07-06:
[printedPerfectWithSlower-BitsWIP-SScrashes.zip](https://github.com/user-attachments/files/16117547/printedPerfectWithSlower-BitsWIP-SScrashes.zip)
sortu that. here itI isI. i have confitrmed thisI problem dors not occur witIh other filIes