supermerill/SuperSlicer #4346: Slicing complex object leads to crash

Icenowy, 2024-07-03:
### What happened?

When slicing dragon_nice_1 from https://www.printables.com/model/210311-articulated-spirit-dragon_-dragon_no-support_easy- , SuperSlicer 2.5.60.0 crashes (this is a regression, 2.5.59.12 doesn't crash).

### Project file & How to reproduce

[empty.3mf.zip](https://github.com/user-attachments/files/16082742/empty.3mf.zip)

As the model is quite big, I am providing a 3mf file with an empty platter here, just add the STL file downloaded.

### Version

2.5.60.0

### Operating system

AOSC OS (distro built SuperSlicer, still testing)

### Printer model

Voron Trident custom size build

Icenowy, 2024-07-03:
A simple fix seems to prevent the crash, see #4347 .

supermerill, 2024-07-03:
your project file is empty.
But anyway, yoru fix reminds me making this modification yesterday, so It should be here (but it isn't, so mistake on my side). So bother recreating it, I believe you.

Icenowy, 2024-07-03:
Ah the empty project file is intended because the STL is 62M.