supermerill/SuperSlicer #4348: Crashing on slicing, stuck on Computing seam visibility

GalvanicGlaze, 2024-07-03:
### What happened?

When slicing the slicer freezes and crashes with a memory error

It happens with one printer profile and not another 

It does not happen in the bis edition but happen in Graph edition 

### Project file & How to reproduce

[image]
[crash_log.txt](https://github.com
[VT.zip](https://github.com/user-attachments/files/16084213/VT.zip)
/user-attachments/files/16084188/crash_log.txt)

### Version

2.5.60.0

### Operating system

Windows 10

### Printer model

Voron Trident

supermerill, 2024-07-03:
maybe same error as #4346 
can you try the nightly when it's compiled in 10 min?
https://github.com/supermerill/SuperSlicer/actions/runs/9777603257

MurseyThings, 2024-07-04:
same issue as above for me w/ graph edition

I'm just trying to slice the voron 2.4 test pieces for tuning and it crashes.

GalvanicGlaze, 2024-07-04:
> can you give me project that hang on slice?

Can you please try this file:
[VT.zip](https://github.com/user-attachments/files/16096167/VT.zip)
this file

PizzaScientist, 2024-07-09:
Do you know what version this bug was introduced in?
I just upgraded from 2.5.59.8 to 2.5.60.0 on Manjaro Linux, and I'm getting these frequent crashes now. I assume it was between these versions?

Tinchus2009, 2024-08-06:
If it is usefull: Im having this issue with the uploaded model, every time, (it has been imposible to slice it).
While new version is released, is there any work around in order for me to be able to slice it?
[conerclamp_slide.zip](https://github.com/user-attachments/files/16511617/conerclamp_slide.zip)

supermerill, 2024-08-06:
yes, disable seam -> use visibility check

Tinchus2009, 2024-08-06:
Oh, I tried that and still gives the error "computing seam visibilities area"

39elbarto, 2024-09-21:
> true. only the nightly https://github.com/supermerill/SuperSlicer/actions/runs/9837348764/artifacts/1677086425 works in the 60 branch.

thank you! Had the same issue with one file. this version doesn't crash!