supermerill/SuperSlicer #4351: superslicer 2.5.59.2 vs 2.5.60.0 flow differences on top fill and solid infill + supporting dense layer percentage

NiiXXiiN, 2024-07-04:
### What happened?

I've been having problems for the last few months with my top layers being underextruded. I was boggled about it but i would manually set 3-5% more flow to the top layers and it got ok at every print. Then i tore down the head of my vcore and thoroughly cleaned it and made sure that i got my extruder lubed and had no partial jams while i put a new nozzle.
After having done all that nothing changed. So i started comparing every version of superslicer i got downloaded after January 2024 to my previous ones and found out that after 2.5.59.2 superslicer started underextruding my top layers. I tried to compensate with top fill (i used to have it to 108%) but nothing worked until i passed 124%.

I also found out that the other versions under extrude solid infill and that the supporting dense layer must be less than 50% infill density.

[image]

### Project file & How to reproduce

[ESP8266 Inclination Base.zip](https://github.com/user-attachments/files/16100311/ESP8266.Inclination.Base.zip)

### Version

2.5.60.0 and 2.5.59.2

### Operating system

windows 11

### Printer model

v-core 3

supermerill, 2024-07-04:
For the first problem (top infill underextrusion):

in 2.5.59.6, i fixed some error about the solid infill, as it was adding two times the encroachment, over-extruding a bit. Now I've added many more tests to ensure there won't be any issue between the computed amount of plastic extruded & the volume to fill.

I looked at what's changed between 2.5.59.2 and 2.5.59.3, and I can't see anything at first glance that may have changed the flow of the infill.

I'm a bit confused. 

I see that you have an extrusion multiplier of 0.88 in your filament. Maybe you're just under extruding everywhere?

> I also found out that the other versions under extrude solid infill and that the supporting dense layer must be less than 50% infill density.

ha, it's computing the spacing with the infill width instead of the bridge one. I think it need to be a bit more intelligent, also to not over-extrude if there's a very thin layer.

NiiXXiiN, 2024-07-04:
For the amount of plastic of infill and top fill.

I used for testing the same print, filament and printer settings. I printed the same stl with 2.5.59.2 and 2.5.60.0 and the result was visible difference both in infill (witch doesn't really matter for me because it doesn't overextrude to the point i lose dimensional accuracy. 2.5.59.2 actually looks better for me because it is truly solid infill - no gaps) and in top fill (2.5.60.0 had quite a lot of linear gaps between straight lines while 2.5.59.2 was just perfect.)

I don't know how to proceed, i mean should i recalibrate all my settings related to flow or wait to see if you find something in the code?

[image]

supermerill, 2024-07-05:
> I don't know how to proceed

Me neither. I can try to print some sample cubes from 59.2 and 59.13.
 * checking manually the amount of plastic requested to be extruded for hte size of the cube (removing the round edges on the sides)
 * printing them, weighting them, to see how they print.
 * report back.