supermerill/SuperSlicer #4358: Crash when generating Bridge Flow Calibration

Arthur-de-Partuur, 2024-07-07:
### What happened?

When I try to run Bride flow calibration, Superslicer 2.5.60 crashes when I hit the _test flow ratio_ button.

### Project file & How to reproduce

Unable to save project file as SS crashes. See clip

https://github.com/supermerill/SuperSlicer/assets/23432540/94f4aa6e-98d8-4cb7-b97d-05d909450aee

### Version

2.5.60

### Operating system

windows 11

### Printer model

RatRig V-Core 3.1 running RatOS 2.1 (Klipper)

Arthur-de-Partuur, 2024-07-07:
Just tried the same with a clean Linux install (so no printer configured yet) and works fine. Could this have something to do with my ini file that I use for config?
[RatRig_borg01.zip](https://github.com/user-attachments/files/16119469/RatRig_borg01.zip)

Arthur-de-Partuur, 2024-07-07:
Above ini file is placed in Vendor dir. Crash seems to be when I select the 0.6mm nozzle printer, in combi with Extrudr ASA-CF. Something in the ASA-CF settings seems to cause the crash.
(also on Linux)

legend069, 2024-07-09:
do you still get the crash using ``2.5.59.13`` ?

Arthur-de-Partuur, 2024-07-09:
> do you still get the crash using `2.5.59.13` ?
> 
> maybe related? #4360

Ill check if the mentioned bug is a similar issue. I am running 2.5.60 which comes after 2.5.59.13, right?

julianjc84, 2024-07-09:
I can replicate the same issue however it only when generating the g-code does it crash.
it does not write a crash_log.txt
seems, the only changes to my configurations are the follow 2 and one of them make its crash.

If i then clear the scene, but keep the changed profile settings and try to slice cube. it quits the app on slicing.

[image]

[image]

Arthur-de-Partuur, 2024-07-10:
And so the plot thickens,... I now get a new error (While I have not changed anything,.... i think.... 
[image]

And I don't think I set something realy strange:
[image]