supermerill/SuperSlicer #4362: Max and min coodinates of a print

Tinchus2009, 2024-07-08:
I ha ve a scriptt that takes min and max xy coordinates of the first layer of a print working on kisslicer. But I like superslicer and I use it more often. Is there any way to get the XY max and min coordinates of the first layer of the print (these should include the coordinates of the brim for example )

If there isnt, I would like to propoise this to be added in a future.

legend069, 2024-07-09:
what are you trying to accomplish with with the min/max values ?
is this what you're looking for?

``PRINT_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} PRINT_MAX={first_layer_print_max[0]},{first_layer_print_max[1]}``

Tinchus2009, 2024-07-10:
I have a script that takes min X max X min Y max Y and adjust the mesh grid . If I dont do it like this, the hole bed grid is made of 100  sensing point, it doesnt make sense to sense the hole bed if Im going to print just a small cube right?

Where did you got those variables? I read docs and couldnt find them

legend069, 2024-07-10:
ahh ok so adaptive mesh? klipper should be adding this to the mainstream but i currently use this my own edited version of [this ](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging)

the variable list should be here https://github.com/supermerill/SuperSlicer/wiki/Macro-&-Variable-list
it might be a little outdated with some recent changes but 90% of it should still be current.