supermerill/SuperSlicer #4365: [FEATURE REQUEST] Z-offset after N-th layer

tecnopolis-ve, 2024-07-10:
Hello, I usually print with PETG a lot, and some tricks that I have gathered over time are that this material does not like to be squished, so a Z-offset of 0.08 mm allows it to lay down nicely, resulting in really good-looking prints. However, this can be really problematic in the first layer, which should be squished on the heated bed (I'm using glass).

I think it would be good and a simple idea to select the Z-offset starting from the N-th layer.

I'm creating this feature request because, to be honest, I don't know if this feature is already present somehow by using a combination of settings. Any thoughts would be appreciated.

Thanks!

[image]

PNG-pyro, 2024-07-11:
You should be able to do this with the 'custom gcode' options under the printer settings. There's a section for layer change gcode, and I think something like 

`{if layer_num = X}G92 Z0.08{endif};`

where 'x' is the layer number you want to shift at should work. You'll likely want to add something to reverse it in the 'end g-code' block to reset for the next print.

Do a bit of research on G92, though, I don't know if it's quite right for this situation or not. But there should definitely be a way to automate this with the custom gcode command settings.

legend069, 2024-07-12:
you can currently do this with 'before_layer_change_gcode'

``{if layer_num == 1 and filament_type == "PETG" } set_zoffset_command {endif}``

foreachthing, 2024-07-12:
Maybe this could help https://github.com/SoftFever/OrcaSlicer/issues/2963#issuecomment-2006343307, too.