supermerill/SuperSlicer #4371: superslicer crashes with particular model orientation

mslmarc, 2024-07-12:
### Version
SuperSlicer_2.5.60.0
Build:     SuperSlicer_2.5.60.0

### Operating system type + version
Windows 11 home Edition
Version	22H2
Installed on	‎1/‎12/‎2022
OS build	22621.3880
Experience	Windows Feature Experience Pack 1000.22700.1020.0

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22621), 64-bit edition
Total RAM size [MB]: 34,012MB
OpenGL installation
GL version:   4.6.0 NVIDIA 546.33
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1650/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### Behavior
Suerslicer crashes when selecting *slice Now*
import model and rotate so that hole is at bottom, select slice now. happens with one or several instances of the same model/stl. 
model should slice as it does on Orca and Cura slicer.  Superslicer acts as normal with any other file. the model will slice any other orientation.
program crashes.
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[reinvented vane.zip](https://github.com/user-attachments/files/16191729/reinvented.vane.zip)

evilphish, 2024-07-16:
I am experiencing the same with a very simple model. If I rotate it onto a particular face slicing crashes the program.

streeriver, 2024-08-14:
Same issue here. Was on 2.5.59.11 and just upgraded to 2.5.60.0, same thing.
[SubaruTrimClip2.zip](https://github.com/user-attachments/files/16617510/SubaruTrimClip2.zip)