supermerill/SuperSlicer #4375: Do not do "External perimeter first" in the first (several) layers

zhou13, 2024-07-13:
**
Is your feature request related to a problem? Please describe.**
I'm always frustrated with perimeter separation in the first several layers in ABS printing when using External perimeter first printing.

Left side: printed with "External perimeter first". I got Perimeter separation consistently.
Right side: printed without "External perimeter first". Perimeter separation problems go away.
[image]

**Describe the solution you'd like**
Add an option to disable "External perimeter first" in the first (several) layers.

**Describe how it would work**
Since we are not doing "External perimeter first" in the first several layers, the perimeter separation problem should go away.

**Describe alternatives you've considered**
I can adjust "better bonding" but it might cause other issues. This is also related to first layer xy compensation but if I disable that I might suffer from the elephant foot issue.

KoiosLabs, 2024-07-14:
I'll add a vote for this, I've got a few models that have this problem, and its kind of annoying to have to pick off or on.  For example the ERCF v2, buffer support arm really struggles with external perimeters first on the first few layers, but otherwise doesnt really even have overhangs, and prints fine after it figures things out :)

legend069, 2024-07-15:
does that model have a chamfer/fillet ?
going to guess that's whats causing the issue more then it being on for the first layer/s

zhou13, 2024-07-15:
Yes, it is the chamfer plus the potential my 0.1mm first layer xy compensation that is causing the issue.

KoiosLabs, 2024-07-18:
same, dont have an issue doing overhangs and such after the first 2-3 layers though, so I see this as a useful configuration option in some cases.  Usually it doesnt cause issues, but certain geometries the filament tension seems to pull it off the thin overlap with the previous layers thanks the the chamfer and angles involved.

zhou13, 2024-07-20:
[image]

Add more photos that the option causes separation on the first layer.