supermerill/SuperSlicer #4378: Total crash while slicing

Fuechsl, 2024-07-16:
### What happened?

crashed while slicing this

### Project file & How to reproduce

[prj crash with cw2_chain_anchor_tilted.3mf.zip](https://github.com/user-attachments/files/16256087/prj.crash.with.cw2_chain_anchor_tilted.3mf.zip)

### Version

2.5.60.0

### Operating system

macos 14.5

### Printer model

Voron V2.4

jueank, 2024-07-26:
[gyro-6x3.zip](https://github.com/user-attachments/files/16391529/gyro-6x3.zip)
Happens with the attached model as well.

shuffleznl, 2024-08-01:
Same for me when generating G-code on Win11. No output/update on console when crashing.

viniavila, 2024-08-10:
I can confirm here. Several files from this project (https://www.printables.com/model/213996-space-shuttle) presents the same error. macOS 14.5, Macbook Air M1 2020.

Problem Report returned after crash as follow:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [33022]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            org.SuperSlicer/
Version:               SuperSlicer SuperSlicer_2.5.60.0_2024-07-01 (SuperSlicer_2.5.60.0_2024-07-01)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-08-09 23:01:24.7712 -0300
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        CC39042A-985A-517A-063F-0A5F92EB6AD4

Sleep/Wake UUID:       EC5DEE4A-750F-4034-89F3-C0295EC14126

Time Awake Since Boot: 120000 seconds
Time Since Wake:       3138 seconds

System Integrity Protection: enabled

Crashed Thread:        9  slic3r_tbb_4

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xfffffffffffffffc
Exception Codes:       0x0000000000000001, 0xfffffffffffffffc

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [33022]

VM Region Info: 0xfffffffffffffffc is not in any region.  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      UNUSED SPACE AT END

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x191de61f4 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x191df8b24 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x191deee34 mach_msg_overwrite + 476
3   libsystem_kernel.dylib        	       0x191de6578 mach_msg + 24
4   CoreFoundation                	       0x191f06680 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x191f04f44 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x191f04434 CFRunLoopRunSpecific + 608
7   HIToolbox                     	       0x19c6a819c RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x19c6a7fd8 ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x19c6a7d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x195763d68 _DPSNextEvent + 660
11  AppKit                        	       0x195f59808 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
12  AppKit                        	       0x19575709c -[NSApplication run] + 476
13  SuperSlicer                   	       0x107147ea0 wxGUIEventLoop::OSXDoRun() + 112
14  SuperSlicer                   	       0x107210b3c wxCFEventLoop::DoRun() + 40
15  SuperSlicer                   	       0x1071998e8 wxEventLoopBase::Run() + 132
16  SuperSlicer                   	       0x1071763b4 wxAppConsoleBase::MainLoop() + 120
17  SuperSlicer                   	       0x10710fbe4 wxApp::OnRun() + 36
18  SuperSlicer                   	       0x1071c17cc wxEntry(int&, wchar_t**) + 164
19  SuperSlicer                   	       0x1054829ec Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 284
20  SuperSlicer                   	       0x10487b1d8 Slic3r::CLI::run(int, char**) + 17448
21  SuperSlicer                   	       0x10487f128 main + 208
22  dyld                          	       0x191a9e0e0 start + 2360

Thread 1:
0   libsystem_pthread.dylib       	       0x191e21d20 start_wqthread + 0
[... 378 lines cut ...]
      "deploymentId" : 240000011
    },
    {
      "rolloutId" : "62699e1ec1ff2978b47f6c3b",
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "652805151bce5442b8d752e1"
      },
      "deploymentId" : 240000031
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c47ab4cc-c9c3-4b5d-a87c-e2433ce02597",
      "experimentId" : "6639bc6ba73d460582162323",
      "deploymentId" : 400000006
    },
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}
```

goeland86, 2024-08-15:
Could this be related to #4396 ?