supermerill/SuperSlicer #4384: Auto Arrange Mis-Functioning

suprchunk, 2024-07-21:
### What happened?

When auto arranging items I expect them to respect the distance I set to keep the separated. But, when I click auto arrange, or press A, items are set on each other.

### Project file & How to reproduce

Import multiple items, auto arrange them by pressing A, or the auto arrange icon in the tool ribbon, have expectations smashed.

### Version

2.5.60.0

### Operating system

Windows 11

### Printer model

Doesn't matter, has nothing to do with the printer selected.

suprchunk, 2024-07-21:
[image]

[image]

legend069, 2024-07-25:
can you please add your project file.

suprchunk, 2024-11-22:
Dang, I'll have to do it again. I didn't see traction for a while so I closed my program, a few hundred times.