supermerill/SuperSlicer #4386: Mouse ears to help with warping

Pandamime100hp, 2024-07-23:
**Is your feature request related to a problem? Please describe.**
No, it is a nice to have to save time in CAD.

**Describe the solution you'd like**
I am finding that when printing with ABS/ASA, anywhere that contains a sharp corner on the print base layer overtime begins to warp. I noticed that "mouse-ears" can hold the corners down or at least help. It is similar to a brim but A) it saves printing time and material (marginal sure), B) saves time in CAD and C) it the print edge much cleaner when you break off the ears. 

**Describe how it would work**
Have 3 variables which generate a circular mouse ear on corners. 
1. mouse_ear_size - the diameter of the mouse ear.
2. mouse_ear_distance - the distance of the mouse ear from the detected corner.
3. mouse_ear_angle_threshold - the angle to which the ears will apply to, say for instance 270° angle (right angle), apply mouse ear but anything less, do not.

**Describe alternatives you've considered**
The alternative is to create mouse ears in CAD during design but it adds extra steps to the overall process if you want to adjust minor changes when it could be done on the slicer.

**Additional context**
3D Slant YouTube channel showing example of mouse ears - https://www.youtube.com/watch?v=MCcFMDv_4eo

macchinette, 2024-07-23:
Hi, do you already have everything you need or am I wrong?
Look at the photo.
[image]

Pandamime100hp, 2024-07-23:
I had absolutely no idea this was already a thing! Thank you @macchinette and apologies for not researching more in advance!

closed 2024-07-23 by Pandamime100hp