supermerill/SuperSlicer #4388: Ironing infill causes underextrusion on next feature

Dycus, 2024-07-25:
**PROBLEM**
With the ironing infill type (or when using traditional ironing), the flow rate is very low. This causes the feature printed after it to be underextruded briefly. This is mainly a problem when the next feature is an external perimeter.

I'm not sure exactly why this happens. Maybe the nozzle pressure gets too low. Or maybe extra plastic is dragged out of the nozzle as it moves, and not enough is going into the nozzle. My extruder is not grinding the filament or skipping. Pressure advance is calibrated.

I understand this is technically an extrusion system issue. However, I think it would be a common problem on many printers and it could probably be fixed in the slicer.

**POSSIBLE SOLUTIONS**
I think there are a few ways this effect could be reduced, but none are perfect.
- Option to extrude a minimum amount of filament into nearest infill area after ironing. Might create a blob if infill has already been printed.
- Option to feed a fixed amount of extra filament right after ironing, before printing the next feature. Easy to implement, but the optimal value for this likely changes with how much time is spent ironing, and will probably always leave a small blob or a gap. But maybe the gap could be reduced.
- Option to print infill first after ironing. If the next feature to be printed is top infill with a single perimeter, then this would probably give poor results.
- Option to print inner perimeter first after ironing. If the next feature has only one perimeter, then the blemish will still show up on the external perimeter.
- Port OrcaSlicer's inner/outer/inner perimeter ordering option. This hides the blemish on the innermost perimeter, while still getting the benefits of printing outer perimeter first. Only works with 3+ perimeters. 

**WORKAROUNDS**
- Enable printing inner perimeter first. This has the same issue as described above
- Enable printing infill first. This has the same issue as described above
- I am still testing this, but setting a higher fill_smooth_distribution helps because the flow rate isn't so low. Reduces blemish but doesn't fix it. I'm currently using 30%, with 50% spacing and 0.5mm top extrusion width.
- Increase ironing speed so extrusion rate is higher. Reduces blemish but doesn't fix it

**ADDITIONAL INFORMATION**
Version: 2.5.59.13
Operating system: Windows 10
I really love the ironing infill for top surfaces, it gives smooth results without increasing print time too much, and helps with dimensional accuracy in Z. I want to make it my default top infill but this blemish is a downside.

[image]

Tinchus2009, 2024-08-15:
Hello. This probably happened to me, not with ASA, but using petg. My symplthoimwas the same as you. The problem to my opinion is the very loss of pressure on the noozle. When ironing finishes, the pressurer in my case was way too low, so the first periimeter on the netx objext was always having a gap because of underextrusion. First solution was to make the inner perimeter to be printed first. Then I looked at the problem and it was moisture: moisture on petg was creatingsagging, so if you extrude by only 15% while ironing, at the last of the ironing movement, there was retraction fort the next object, and in the second it takes to move the printhead, oozing was hapening because of moisture, so the very few pressure and material inside nozzle was lost, causing the underextrusion.
Im not saying you have moisture because I dont see that in the picture, but you could try checking if you are not printing external perimeter first, in that case Im pretty sure printing the internal perimeter first (and so acting as a prime) will solve the problem

CuredPrusa, 2024-10-20:
Hi!
This is very normal and not related to slicer. Of course, software could be tweaked to compensate anything you like, but no one want's to handle the implementation of such features (as far as I know). I often do ironing. Not only that I select inner perimeter first on next layer, I also sometimes call for infill first if infill is big enough to build up the pressure and continue to outer perimeter first. It works for me nicely.