supermerill/SuperSlicer #4389: 2.5.60.0 -- Freezes and closes with no error message

amsmith525, 2024-08-01:
### Version
Version 2.5.60.0

### Operating system type + version
Edition	Windows 11 Pro
Version	23H2
Installed on	‎2/‎25/‎2023
OS build	22631.3958
Experience	Windows Feature Experience Pack 1000.22700.1026.0

### Behavior
Try to load attached 3MF file in 2.5.60.0 -- freezes and closes with no error. Try to open 3MF in 2.5.59.8 -- works fine.

[image]

#### Project File (.3MF) where problem occurs
[3MF file.zip](https://github.com/user-attachments/files/16449774/3MF.file.zip)

roykrikke, 2024-08-02:
I have the same with on other file. See below for the file.
[teslavent.zip](https://github.com/user-attachments/files/16469755/teslavent.zip)

JanB97, 2024-08-04:
Try the newest working nightly under actions. It fixed some of those issues. Mine crashed with the release 2.5.60 as well, but doesn't with the nightly.

amsmith525, 2024-08-04:
> I have the same with on other file. See below for the file. [teslavent.zip](https://github.com/user-attachments/files/16469755/teslavent.zip)

Strange - I tried opening your file and it does not crash