supermerill/SuperSlicer #4396: Failed to load STL file

vladi552, 2024-08-07:
### What happened?

The file included cannot be sliced. Need investigation.

### Project file & How to reproduce

Direct load and slice.

### Version

2.5.60.0

### Operating system

win11-64

### Printer model

Tevo Black Widow

vladi552, 2024-08-07:
[File.zip](https://github.com/user-attachments/files/16530140/File.zip)

foreachthing, 2024-08-14:
@vladi552 it just sliced fine:
[image]

Maybe you should upload your full .3mf (zipped).

Win 11, 2.5.60 (titlebar says 2.5.59....??)

vladi552, 2024-08-14:
Ok, uploaded.
[Uploading Full.zip…]()

goeland86, 2024-08-15:
> Win 11, 2.5.60 (titlebar says 2.5.59....??)

The titlebar bug is noted in the release notes for 2.5.60 BTW.