supermerill/SuperSlicer #4403: "SEAM" on vase mode

Tinchus2009, 2024-08-15:
### What happened?

Im not sure if this is  abug. Im printing simple cone shaped objects and on all of them I get a sort of seam. It is not technically speaking a seam, there is no retraction made of course, this "seam" is made by s slight movement of the perimeter at the end of every turn to increase diameter I guess, but I dont understand why this round perimeter is not "closed"  at the end of perimter creation.
I have tried doing the same on cura spiral mode and somehow this efffect is covered / not present.
I have attached pictures, stl and the project. The picture might look like "it is nothing" but the picures hiddes the deffect a lot, in real life if much more visible.

Also I think there is another bug on vase mode: vase mode should not make retractions. If you activate vase mode and in the printer config you have "retraction on layer change", this setting will take affect, seems a retraction is made and on every turn you can see a retraction ahppening. I think this setting should have be deactivated together with the other settings when you activate vase mode.
(this setting is not active and not happening on my report)

Also a question: is there any way to have bottom layers on vase mode?

Thanks in advance for the help

[image]
[image]

### Project file & How to reproduce

[vasemodecone.zip](https://github.com/user-attachments/files/16624937/vasemodecone.zip)

### Version

2.5.60.0

### Operating system

Win11

### Printer model

Custom FDM cartesian

Dougwar, 2024-08-16:
Thats an old and know  issue
look here

https://github.com/supermerill/SuperSlicer/issues/2047#top

Tinchus2009, 2024-08-18:
Ok, thanks for clarification. In IMHO vase mode is a function very much used and usefull, having this seam is like not having a powerfoll tool. Dont take this comment in a negaqtive way please, just an opinion. I have checked and on slic3r 1.3 stable version, the vase mode works perfect (I did the test because slic3r was the slicer I used from the very beggining and the one before switching to superslicer). May be go bak to that "old" vase mode algorithm?